dekuNukem/duckyPad #151: Duckypad won't connect

harkfujow, 2024-02-12:
My Duckypad has stopped working. The screen says "Please insert SD Card" and the firmware is V1.3.0. I have the exact same issue as in [#149 ](https://github.com/dekuNukem/duckyPad/issues/149l). I think the SD card is dead. I tried reformatting it, but it failed in Windows and it's detected at all in Linux.  Replacing it with a new, and freshly formatted card didn't work. On Linux (Manjaro), it might as well not be connected. On Windows 10, the card is detected in the configuration app, but the Duckypad itself remains inaccessible. I've replaced the cable and also connected the Duckpad directly to the Linux PC and the Windows PC. No dice. I'm now at a loss as to how to proceed.

dekuNukem, 2024-02-12:
Sorry to hear about the issue, is the duckypad not recognising the new SD card? What does the screen say with the new card inserted?

harkfujow, 2024-02-12:
It says the same thing as if the old card or no card was inserted: "Please insert SD Card. For instructions, see duckyPad.com. FW V1.3.0"

dekuNukem, 2024-02-12:
Make sure it is formatted in FAT32, see the "I want to use my own SD card" section here: https://github.com/dekuNukem/duckyPad/blob/master/troubleshooting.md#i-want-to-use-my-own-sd-card

duckypad might be a bit picky as well, try a few different card if you have some.

let me know how it goes!

harkfujow, 2024-02-12:
Well, I can't format the spare card. There are two partitions on the 128GB card. My options are NTFS and exFAT on the larger partition and FAT16 on the smaller one. Looks like I'm buying a new card.

harkfujow, 2024-02-13:
So, nothing is working. I tried manually re-flashing the firmware. I was able to do so. Unfortunately, it made no difference. The duckypad is still stuck on "Please Insert SD Card." I'm at a loss and thinking the duckypad is irreparably broken.

dekuNukem, 2024-02-13:
I would suggest try a few different SD cards first, lower capacity (under 32GB), one partition in FAT/FAT32. If that doesn't work, I'm happy to send you a new circuit board for free!

harkfujow, 2024-02-13:
Unfortunately, I have only one spare 32GB card, and that is the original card from my pi-hole. I kept it as backup in case my pi-hole SSD bites the dust. The card that came with the duckypad appears to be dead. It doesn't show up when I plug the adaptors (I have a few) into the PC. My other spare cards are 128GB+. Looks like I'm headed to Amazon for lower capacity cards.

dekuNukem, 2024-02-13:
sure, go for 8 or 16gb if its not too expensive. if that works i can reimburse you the cost. if still doesn’t work, i can send you a new circuit board.

harkfujow, 2024-02-20:
No luck. I tried multiple SD cards to no avail. I think it's dead.

dekuNukem, 2024-02-20:
well, I can send you a new PCB then! Let me know your tindie order number. Same address as last time?

harkfujow, 2024-02-20:
It was order #421914. No change to my address. Thanks!

dekuNukem, 2024-02-20:
I'll send it out later today!

harkfujow, 2024-02-28:
I received the new board and everything seems to be working. I copied one of my backups onto the SD card and we're good to go. I will say that the LEDs seem brighter. 

Thanks for all of your help.

closed 2024-02-28 by harkfujow