dekuNukem/duckyPad #156: PC does not see duckypad as an input device.

poonmelissa, 2024-06-19:
Hi, I was trying to use Duckypad on my new Laptop and it only sees Duckypad as other devices, not an input device. 
Under device manager, Duckypad is under Universal Serial Bus Devices, can not find STM32 Bootloader. 

Not sure what is missing for the Duckpad to work.
Please help.

Thanks
Melissa

dekuNukem, 2024-06-19:
Hi Melissa! Does it work on other PCs? Try a few different ports and cables. Also does the RGB backlight change when you press a key?

poonmelissa, 2024-06-19:
I have not been updating the firmware since I brought it years ago, V0 19 1
Tried to update but not successful with DfuSe , STM32 ....
I followed what was on your post but not successful

dekuNukem, 2024-06-19:
i do think it's a good idea to update it to the latest firmware. the guide is here: 

https://github.com/dekuNukem/duckyPad/blob/master/firmware_updates_and_version_history.md

When you plug it in DFU mode, the light and screen should stay off, is that the case?

if so, what shows up in the device manager?

poonmelissa, 2024-06-19:
Not showing up on the Device Manager.

poonmelissa, 2024-06-19:
[image]

poonmelissa, 2024-06-19:
I have a USB Card reader for the SD card, Can you send me updated files and I tried to upload to the card and  let me know what am I missing on my laptop to it to recognise the duckypad as an input device,
I am not sure if I messed up anything when I tried to use DfuSE Demo to update firmware and it got frozen not responding and I had to restart to get out of that.

dekuNukem, 2024-06-19:
it does show up, so that means it is working.

Here are the sample profiles:

https://github.com/dekuNukem/duckyPad/raw/master/sample_profiles.zip?raw=1

mount the SD card on PC, copy files off if you want, do a format in FAT32, unzip and copy the sample profiles to the card, see if it works better.

if Dfuse demo doesn't work on your laptop, maybe try a different computer if possible, also make sure to enter DFU mode properly, the screen and RGB LED should be off.

poonmelissa, 2024-06-20:
did all that and was sure it si in DFU Mode, led is off. After hitting upgrade the operation duration started to count but no movement on the progress bar

dekuNukem, 2024-06-20:
ah, I've heard that before, usually happens with AMD motherboards.

Try plug it in through a cheap USB hub, that usually fixes it.