dekuNukem/duckyPad #158: Duckypad worked two years ago - but no longer

dabrams, 2024-07-21:
I last used my Duckypad in 2022.  I hooked it up again today and it just displays "Please insert SD card..."  My USB manager shows it as USB device but neither the 0.13.3 or 1.6.3 configurator can connect.  I tired to flash the firmware - it goes into STM Bootloader mode just fine - but the "Available DFU Devices" dropdown is empty.  Windows says I have the latest driver.  I have tried three different USB ports and three different USB cables.  I also rewrote the SD card from my last good back up and also tried rewriting the SD card with the original data.  Not sure what to try next?

David  
[image]
[image]

dekuNukem, 2024-07-21:
there has been a lot of updates since 2022, so it is a good idea to update to the latest firmware!

Have you tried reinstalling the driver? Take a look at this guide: https://github.com/dekuNukem/daytripper/blob/master/dfu_driver_install.md

dabrams, 2024-07-22:
> Have you tried reinstalling the driver? Take a look at this guide: https://github.com/dekuNukem/daytripper/blob/master/dfu_driver_install.md

Yes, I removed the driver and reinstalled it as well.

dabrams, 2024-07-22:
Ok, I was able to update the firmware to 1.3.0 but still not able to get it working. The duckyPad is seen as a USB Input Device by Windows so it is there. 

I tried a different micro SD card but that made no difference.  It acts as though it cannot see the SD card and just sits there with the "Insert SD Card" message. The configuration utility (V1.6.3) never connects (it doesn't time out but the connect button stays down and nothing happens) but if I remove the SD card it pops up a "duckyPad not found" error dialog" so the board is aware when the SD card is removed.  My computer gives a USB dis/connect beep when I remove the SD card as well.  

Anything else I can try?

[image]

[image]

dekuNukem, 2024-07-22:
that's good progress! Take a look at the `I want to use my own SD card` section of this guide: https://github.com/dekuNukem/duckyPad/blob/master/troubleshooting.md#i-want-to-use-my-own-sd-card

Maybe try a few different card too! Let me know how it goes

dabrams, 2024-07-23:
I have tried several SD cards with no luck.  Does the error message mean the software does not detect an SD card or that the card does not have the correct files on it/cannot read them ?

dekuNukem, 2024-07-23:
that message means the firmware can't detect a valid SD card. A different message is shown if it detects the card but the card is empty.

i would suggest try some smaller capacity cards (under 16GB), and make sure they are formatted in FAT32.

Also try it on a different PC if possible, some AMD motherboards have USB compatibility issues.

Maybe also clean out the slot with some compressed air, something might have gotten inside.

If nothing works, then it might be a hardware issue, and I'm happy to send you a new PCB!

dabrams, 2024-07-24:
Thanks.   I cleaned the SD slot with contact cleaner a few days ago and I just blew it out with compressed air with no change.  It does not seem to be PC motherboard related, since it occurs also when plugged into a USB C power adapter.  (I also tried it with my surface which is an Intel microprocessor.)  I will try to find a smaller SD card (everything I have tried so far is 32g or larger - but they are formatted w/ FAT32) and let you know how it goes.

dabrams, 2024-07-24:
Thanks. I cleaned the SD slot with contact cleaner a few days ago and I just blew it out with compressed air with no change. (I looked in SD slot with a light and magnifier and the contacts look clear.) It does not seem to be PC motherboard related, since it occurs also when plugged into a USB C power adapter. 

 

I found a 2Gig microSD, formatted it for FAT32 and copied the sample_profiles to it, then tried it in my Surface which has a Intel i7 processor.  Still shows the “Insert SD” message and the config utility cannot connect to it. I also reflashed the firmware with verify checked just to be sure.

 

Happy to pay for a new PCB, I just would prefer not to buy a whole kit again.

 

Just so you know, the reset button is broken but I checked and the pullup on the reset side (NRST) is at 3.3V (and I would assume the error message would not display if reset was asserted).

 

David

dekuNukem, 2024-07-24:
Looks like a hardware issue then!

I'm happy to send you a new PCB for free, no need to pay! Let me know your Tindie order number so I can look it up.

dabrams, 2024-07-24:
Well, that doesn’t seem fair.  I bought it more than two years ago. I should pay something.

dekuNukem, 2024-07-25:
don't worry about it! it's my hobby project so I dont have to nickel and dime the customers! let me know the order number and I'll send you a new PCB!

dabrams, 2024-07-26:
#304973 <https://www.tindie.com/orders/purchases/304973/> 

04 Jan 2022

DuckyPad: Do-It-All Mechanical Macropad <https://www.tindie.com/products/dekunukem/duckypad-do-it-all-mechanical-macropad/> 

Review Now <https://www.tindie.com/reviews/create/dekunukem/duckypad-do-it-all-mechanical-macropad/> 

Shipped
2022/01/06
Royal Mail (No Tracking)

$98.96

Ok, Thanks.

dekuNukem, 2024-07-26:
I have dropped it off! It should arrive in the usual time!

closed 2024-07-26 by dekuNukem

dabrams, 2024-07-26:
Thanks – do you want the  old one back?