dekuNukem/duckyPad #160: Please Insert SD Card

fisherbt, 2024-08-12:
My DuckyPad just started showing and Error to Please Insert SD Card, but I tested the SD card on another DuckyPad on the same computer and it works fine. Trying the SD card from the other DuckyPad in the suspect DuckyPad does resolve. The issue. I'm on the latest firmware 1.3.0. Is there a hard reset or anything I can do to get this working again?

dekuNukem, 2024-08-12:
Hi there! I would try giving the SD card a full format in FAT32 and copy the files back. 

You can also try your own SD card if you have a spare. 

See `I want to use my own SD card!` section of this guide: https://github.com/dekuNukem/duckyPad/blob/master/troubleshooting.md#i-want-to-use-my-own-sd-card

fisherbt, 2024-08-12:
Dang, okay, yep. That worked. Thanks for the help.

closed 2024-08-12 by dekuNukem

fisherbt, 2024-08-15:
Unfortunately, this issue is back.  I've even purchased a new SD card formatted it and added my files to it. It will work for a little while after the card has been formatted and then it will go back to saying "Please Insert SD Card" it seems like something beyond the SD card. What else can I try?

dekuNukem, 2024-08-15:
hmm thats unusual. how many times has that happened? maybe I can send you a new PCB if it keeps happening.

fisherbt, 2024-08-15:
It's been happening a couple times per day for the last few days, since I opened the original issue.

dekuNukem, 2024-08-15:
no problem, I'll send you a new PCB with some extra SD cards for free! Let me know your Tindie order number.

fisherbt, 2024-08-15:
Thanks, so much! My order number is 441920.

dekuNukem, 2024-08-15:
Got it! Same address as before?

fisherbt, 2024-08-15:
Yes, thanks. I appreciate you handling this.

dekuNukem, 2024-08-16:
I have dropped it off! it should arrive in the usual time.

fisherbt, 2024-08-16:
Thank you so much for taking care of this.