dekuNukem/duckyPad #161: Feature Request: Print variables as text/strings

coreman14, 2024-08-12:
Right now when you type out text through `STRING` or the `OLED_PRINT`, it has to be predefined in a constant or by the script itself.

Because of this, if you want to have more than one options for text output, you need to add if statements.

I purpose that to put a dent in the above problem, a new command is added that converts an int to an ascii character which is then outputted.

Something along the lines of `CHAR`, `CHARLN` and for the oled screen `OLED_CHAR`.

Which would allow for
```
VAR $i = 0
$_RANDOM_MIN = 65
$_RANDOM_MAX = 90

WHILE $i < 5
    STRING Random Letter: 
    CHARLN $_RANDOM_INT
    $i = $i + 1
END_WHILE
```

dekuNukem, 2024-08-13:
Did you mean printing a variable in STRING, STRINGLN, and OLED_PRINT?

Pretty sure they can already do that. Try 

`STRINGLN Here's a random number: $_RANDOM_INT!`

coreman14, 2024-08-13:
Yes, but it prints the variable as a number. I would like to ability to letters without having to define every possibility. So I would like to be able to do this
```
VAR $i = 0

WHILE $i < 26
    CHAR 65 + $i
    $i = $i + 1
END_WHILE
```
Where the output would be `ABCDE...`