dekuNukem/duckyPad #163: Feature Request: Ability to revert to a backup version via DuckyPad Configurator

richardgaunt, 2024-09-10:
At the moment the only way to restore a backup is to connect the SD micro card to your computer and then manually copy across the backup. This is a problem for me as I don't have a SD Card reader.

I would like to know whether it would be possible to implement this via the application.

Optimally it would be great to see changes between the different versions even if that means the actual macro are stored as text strings of their actual commands rather than the number sequences they are currently (not sure what format).

dekuNukem, 2024-09-10:
A workaround is:

* Connect to duckyPad on the configurator
* Press `x` button and delete all the existing profiles
* Press `import` button, go to backup folder, and select the backup you want
* It will load it up, then you can write it back

An SD card reader should be included in the package as well, maybe see if its still there?

richardgaunt, 2024-09-11:
Nice thanks that is a great workaround appreciate it

closed 2026-02-25 by dekuNukem