dekuNukem/duckyPad #165: Question about special keys

iguanaonmystack, 2024-11-16:
Hi, I'm looking into getting a duckypad but I wanted to make sure it supported key combinations where one of the keys is assigned to Shift, Ctrl, or Alt. From the looks of the docs it seems to, since SHIFT, CTRL, etc are duckyscript commands, but I've had trouble with eg Windows PowerToys's keyboard remapper on a USB numpad, where key combinations don't work, even with one key set to Shift.

So if I had a DuckyPad with Key 1 set to Ctrl, and Key 2 set to C, would I be able to hold down Key 1 and press Key 2 to send Ctrl+C, as if I'd pressed those two keys on a regular keyboard? Additionally, would holding down Key 1 and clicking with my mouse (or pen tablet's stylus) provide the OS with a Ctrl+click?

I ask because I'm considering using the DuckyPad with art software, but given how many keyboard shortcuts are there, I need more actions than just the number of keys on the pad. Having three of them providing Ctrl, Shift, and Alt, would more than triple the amount of use I'd get out of the rest of the keys (eg, C is lassoo, while Shift+C is magic wand select, Ctrl+C is copy, potentially Ctrl+Shift+C is something else still...). Additionally, Ctrl/Shift plus mouse drag provides zoom-related features.

Thank you!

dekuNukem, 2024-11-16:
Hi there! Thanks for getting in touch! What you described is indeed possible! 

There is a specific `EMUK` command that makes a key behave like a regular keyboard key. 

So you can assign `EMUK CTRL` to one key, and `EMUK c` to another. And it will do what you described.

Feel free to read more about it in the EMUK section here: https://github.com/dekuNukem/duckyPad/blob/master/duckyscript_info.md#emuk

iguanaonmystack, 2024-11-17:
Thank you for your reply! That's great to know :D

closed 2024-11-17 by iguanaonmystack