dekuNukem/duckyPad #166: Hotkey combination from PC keyboard to change profiles

pingusix, 2024-12-28:
Hi, I've managed to snap off the buttons on my duckypad (crammed in awkwardly into a small pouch on a work trip). 
Is there any way to assign a hotkey combination on my laptop keyboard to cycle through keyboard profiles? Alternatively could anyone point me at a code sample to try to knock something up? Cheers

dekuNukem, 2024-12-28:
Hi there! You can use `PREV_PROFILE` and `NEXT_PROFILE` command to cycle through profiles, assign to one of the mechanical switches and it should work just like before!

pingusix, 2024-12-28:
Thanks! I did try that but when when I moved to the next/prev profile it obviously didn't have the binding any more so I'd need to add it to all profiles which is a loss of two keys on each one which is a shame. I may have a look at the profile auto-switcher code otherwise I may just have to dust off the soldering iron and get a couple of micro-switches as a worst case.