dekuNukem/duckyPad #169: No/unstable connection with FW 2.0.0 for DP 2020

wellmeyer, 2025-04-25:
After flashing 2.0.0 onto DP 2020 it looks fine and working on the DP screen, but there are major connection problems.
Directly connected to a Windows 11 laptop the new configurator (3.1.0) found it one time and was able to load the profiles.
Also, the Autoprofile 1.0.0 works from time to time, but pressing keys on DP 2020 has no effect.

When connected via a dock nothing works. DP 2020 shows the profiles on the display, but both configurator and autoprofile don't find it and also here pressing keys has no effect.

After I flashed back to 1.3.3 everything again works as expected, directly connected and also via the same dock as above.

The DP Pro works fine with the new FW and the the new programs.

Would be great if this could be solved some when. Thank you.

dekuNukem, 2025-04-25:
that's a bit strange, the 2.0.0 firmware is bascially a total re-write with most of the code backported from the DPP, and there are a lot of differences under the hood. 

Ideally you should load the profile in the latest configurator and then write it back for everything to work. maybe try it on a different PC? I hope it's not like that on every computer, it seems to work fine on my 2 PCs and 1 mac.

wellmeyer, 2025-04-27:
Hi dekuNukem,
I have an update for you after trying again, paying attention to load and save (after removing EMUKs) in configurator 3.1.0.
I've (probably) the same results as before, but now more detailed:
It works connected via USB C to USB C cable, both directly and via the dock (after switching to dock (or between TB/USB C ports on the dock), sometimes I have to press reset to get it working).
It works directly connected via USB A to USB C cable to the laptop.
It doesn't work connected via USB A to USB C cable to the dock (same cable).

Please note that I tried different USB A ports on the dock and that with FW 1.3.3 it works flawlessly with the same cable and same port and did it from the beginning. So it's definitely FW 2.0.0 that makes the difference.
BTW, DPP is working fine with same USB A to USB C cable at the same port on the dock.
It's a Lenovo TB4 dock (40B00135).

Now that DP 2020 works connected to the dock, it's less urgent, but it would be great getting back the original cable configuration eventually, just because of my personal cable and port availability.

That said, still I want to thank you for the great work that you did with both DPs!

Have a nice weekend.