dekuNukem/duckyPad #170: Issues with FW >= 2.x and new/dedicated configurator app

4ch1m, 2025-07-14:
Hi there,

I'm a happy duckyPad user for about four years now. It's an indispensable time-saver. Thanks! :+1:

In the past, all firmware (and configurator app) updates worked without a flaw.

However, with the latest update(s) my duckyPad refuses to work. :cry:

Until recently I was running FW `1.2.0` and the ["old" configurator app](https://github.com/dekuNukem/duckyPad/tree/master/resources/duckyscript%201%20fw%20and%20app/duckyscript1) from the main repo.

Once I found out that the configurator app now moved to a dedicated [repo](https://github.com/duckyPad/duckyPad-Configurator), I just gave it a try.

* I followed the provided [instructions for Linux](https://dekunukem.github.io/duckyPad-Pro/doc/linux_macos_notes.html); and successfully started the new app. :heavy_check_mark:
* A connection to the duckyPad seems to have been established; and I was informed that I had to upgrade to firmware greater/equal `2.0` in order to use the new app. ✔️
* So, as usual, I downloaded the latest firmware ([2.3.2](https://github.com/dekuNukem/duckyPad/blob/master/firmware/duckypad_v2.3.2.dfu)); and flashed it to the duckyPad. No errors. :heavy_check_mark:

After turning the duckyPad off/on the display stated that there are no profiles present.
And after closing/re-opening the new configurator app (and clicking the connect button) I just got this message on the display:

```
FATAL ERROR: 10
```

Using the "old" configurator app with the new firmware also didn't work. (It simply wouldn't connect to the duckyPad anymore.)

I've tried several things (completely wiping/initializing the sd-card, etc.).
Without success.

After downgrading the firmware to `1.3.3` and using the "old" configurator everything is working back again. :relieved:

I hope these issues with the new FW/configurator can be resolved.
Let me know if there's anything I can provide for debugging.

Thanks.

dekuNukem, 2025-07-14:
Hi there! Sorry to hear about the issue. Please try out this guide and let me know how it goes!

https://dekunukem.github.io/duckyPad/starting_fresh.html

4ch1m, 2025-07-17:
Thanks for the response.

Unfortunately the issues still remain.

I closely followed the instructions ("Starting Fresh with duckyPad (2020)"):

* unplugged the duckypad
* removed the SD card
* formatted another fresh SD card with FAT32; copied all files from `sample_profiles.zip` to it
* updated the firmware (as described to) v2.3.2
* put the SD card into the duckypad
* unplugged/replugged the duckypad

As for these steps, imho everything worked just fine.
The duckypad boots; all profiles appear on the screen and are switchable/usable.

However, the new configurator-app seems to have issues.

I'm using this (custom) `run.sh` script to directly run the app:

```
#!/bin/bash
cd "$(dirname $(realpath "${0}"))" && \
  python3 -m venv .venv && \
  source .venv/bin/activate && \
  pip install -r src/requirements.txt && \
DUCKYPAD_UI_SCALE=2 \
python src/duckypad_config.py
```

The app starts without a problem.

But once I hit the "CONNECT" button, I get this error dialog:

```
Empty HID response!

Try a different USB Port Or a USB Hub
```

The stdout shows this:

```
user selected: {'fw_version': '2.3.2', 'dp_model': 20, 'serial': '********', 'hid_path': b'1-8:1.0', 'hid_msg': [4, 0, 0, 2, 3, 2, 20, 30, 111, 40, 120, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]}
DP20 this: (2, 3, 2) 
remote: (2, 3, 2)
```

And the duckypad-screen shows this:
```
File Access Mode
```

Note:
When downgrading the firmware to < 2.x (and using the "old" configurator app) everything works again; with the same hardware-setup.

Any clues?

Can anyone confirm that the "classic duckypad" still works with FW 2.3.2 and the new configurator app **on Linux**?